WROMagnet/Magnet-Robots #20: Problema con GITHUB

WROMagnet, 2024-05-02:
### Tuvimos un problema con GITHUB ya que @Karlesolano @LuisriosWRO3 y yo @DanielRomanRodriguez no sabiamos como utilizar la plataforma Github y no teniamos nada pero buscamos unos papeles que nos entrego Janice y los leimos y pudimos hacer unas par de cosas pero aun asi faltan mas cosa de agregar a Github pero no sabemos como hacerlo.

WROMagnet, 2024-05-02:
Actualmente hoy 1 de Mayo de 2024 no sabemos aun como agregar las cosa ya que no sabemos utilizar bien la plataforma pero intentaremos ver como solucionarlo.

DanielRomanRodriguez, 2024-05-02:
- **Mañana leeremos otra vez con calma los pápales de GitHub a ver si podemos encontrar algo que nos ayude**

DanielRomanRodriguez, 2024-05-08:
**Hoy 8 de Mayo e logrado entender un poco más el GotHub así que podríamos decir que solucionamos este problema aunque aún nos falta el Code pero veremos cómo se hace después.**

WROMagnet, 2024-07-07:
**Hoy 7 de julio de 2024 yo @DanielRomanRodriguez estoy entendiendo mas la libreta de ingenieria me falta otras cosas pero ya se usar github asi que ya doy concluido esta problema.**

closed 2024-07-07 by WROMagnet